RPi-Distro/raspberrypi-ui-mods #1: LXPanel disappeared permanently after removing shortcuts

cefn, 2015-06-15:
Hope this is the right place for the bug.

I just booted a completely new Noobs image, ran a dist-upgrade and removed three applet icons from the LXPanel Application Menu. The panel then disappeared altogether and I have not seen it since. 

Fortunately I already had a terminal open, so I was able to attempt some things. Killing the lxpanel process and relaunching it with the same command line as the original process (same profile) didn't cause it to reappear. Rebooting the machine also didn't cause it to reappear. 

Without LXPanel visible, there are no obvious means of running applications at all, not even LXPanel.

This is an extremely serious bug as changing the icons in the Application Menu it's a likely first step for someone launching the Raspbian desktop and it has been immediately and completely fatal to the use of the desktop, meaning I will probably be forced to reinstall NOOBS.

WORKAROUND

I have been able to right-click on the desktop, then choose in one of the tabs an option like 'make Window Manager menus appear when desktop is clicked' (this original set of tabs is now inaccessible). This config change caused a new menu to appear on desktop-click, which includes a 'Terminal Emulator' launch entry.

Running the Terminal Emulator, I can see through ps aux | grep lxpanel that indeed lxpanel --profile LXDE-pi is running, but remains invisible. 

Launching the regular lxpanel I get a visible menu at the bottom which is usable, but overall this is very problematic and means there is something seriously wrong with the behaviour of the lxpanel in its normal profile configuration, at least on first boot.

spl237, 2015-06-15:
Can you please give some more detail as to exactly what you did to provoke the problem, and I can then investigate?

Specifically, which three icons did you remove from the menu, and how did you remove them?

cefn, 2015-06-15:
I right-clicked on the Web icon, and selected Panel Settings
Under PanelApplets I found an entry 'Application Launch Bar' and clicked 'Preferences'
I then clicked 'Remove' against Epiphany Web Browser, Mathematica and Wolfram in some order or another. 

Alongside this I was probably launching and closing browser windows and apt-get installing python-software-properties.

The result seems to have been that the panel file in the proper place in the .config tree simply disappeared (perhaps was deleted for rewriting, and then rewriting failed or something - potentially a race condition around background filesystem activity with apt?).

I have found a file at ~/oldconffiles/lxpanel/LXDE-pi/panels/panel which appears to contain the original config. This file is simply not there in .config/lxpanel/LXDE-pi/panels

Luckily, moving the copy of the panel file from the oldconffiles tree to my user account's .config tree brought the LXPanel back, so my Raspbian is usable again.

I can't seem to recreate the issue right now by following the same steps.

XECDesign, 2015-10-04:
Closing due to inactivity and inability to replicate.

closed 2015-10-04 by XECDesign