RPi-Distro/raspberrypi-ui-mods #3: Upgrade message box too large to dismiss on small TFTs

scruss, 2016-06-16:
The “Your Raspbian system has been upgraded …” box set up at [raspberrypi-ui-mods/raspberrypi-ui-mods.postinst at master · RPi-Distro/raspberrypi-ui-mods](https://github.com/RPi-Distro/raspberrypi-ui-mods/blob/master/debian/raspberrypi-ui-mods.postinst#L55 "raspberrypi-ui-mods/raspberrypi-ui-mods.postinst at master · RPi-Distro/raspberrypi-ui-mods") line 55 keeps coming back on screens smaller than the zenity window. One has to manually delete the ~/.config/autostart/pi-conf-backup.desktop file to make it stop.

A minor issue, I know, but one that affected me.

spl237, 2016-06-16:
Sounds like that's a Zenity bug rather than anything we can fix, but I'll have a look. We might have to use something other than Zenity to show the message.

waveform80, 2018-02-01:
I'm guessing that instead of clicking "OK" in that dialog box (which would exit zenity with 0, permitting the `&& rm` after it to run), you're instead closing the dialog box with the "X" or pressing Alt+F4 (which exits zenity with 1 prevent the `rm` from running) possibly because the dialog can be quite large on a small screen and the "OK" might be falling off the right.

Probably a couple of things can be done about this: don't use `&&` to trigger the `rm`, just use `;` instead. And it might be worth "thinning out" the text displayed with zenity (or throwing in some more line-breaks) so that on tiny/low-res screens the OK button doesn't fall off the right.

texadactyl, 2018-07-11:
Is there a parameter in some config file available to mask out this message?  It is simply a royal pain and covers up the display after a boot.

spl237, 2018-07-11:
You can close the window, however big it is, by just pressing Enter while it is on display. That should allow the subsequent rm to run and remove the autostart entry which shows it.

We are considering possible alternatives to this at the moment, but the Debian install system does not make it easy to update configuration files when we need to.

closed 2018-07-11 by spl237

texadactyl, 2018-07-11:
Cannot press Enter; there is no keyboard or mouse.  However, I am not stuck.