RPi-Distro/raspberrypi-ui-mods #5: There's no easy way to connect to hidden SSID's in the GUI.

mintslice, 2016-10-11:
The Network Preferences doesn't allow you to specify a hidden SSID, making connecting very difficult.

After selecting SSID, the list only includes "non hidden" SSID's (or nothing if there are no other SSIDs) and it's not possible to specify a SSID if you're trying to connect to a hidden network.

There should be an option to select "Other network" and then specify the SSID manually.

XECDesign, 2016-10-11:
@spl237

spl237, 2016-11-25:
We use dhcpcd5 for networking, and a modified version of dhcpcd-gtk as the UI. It doesn't offer this functionality, so unless the author adds it, I'm afraid it won't be available in the near future.

Hiding of SSIDs is largely pointless anyway - it doesn't increase security against hacking to any significant effect, as the tools used by hackers can see networks whether or not they are broadcasting SSID. I gave up bothering with hiding SSID on my home wifi years ago...

closed 2016-11-25 by spl237