RPi-Distro/raspberrypi-ui-mods #7: raspberrypi-ui-mods update assumes the existence of user "pi", unable to update package

SarenT, 2016-11-28:
Hi,

I have recently updates some packages via aptitude and I have got the following error during the update process:

    /home/pi/.config/gtk-3.0/gtk.css
    chown: invalid user: ‘pi:pi’
    dpkg: error processing package raspberrypi-ui-mods (--configure):
    subprocess installed post-installation script returned error exit status 1
    Setting up raspi-config (20161114) ...
    Processing triggers for libc-bin (2.19-18+deb8u6) ...
    Errors were encountered while processing:
    raspberrypi-ui-mods
    E: Sub-process /usr/bin/dpkg returned an error code (1)
    Failed to perform requested operation on package. Trying to recover:
    Setting up raspberrypi-ui-mods (1.20161115) ...
    Backing up old config files...

I don't have a user called pi anymore (due to security reasons). And I guess the package should not assume the existence of such a user anyway.
Please note that, I have unlocked aptitude and started again, it doesn't show any issues or updates related to the package.

XECDesign, 2016-11-28:
There is currently a test for the existence of /home/pi, but not for the existence of the pi user itself (this will be fixed in the next version or the one after). As a workaround, you could move /home/pi to /home/pi.bak.

We are trying to take out references to the pi user wherever possible, but that's not always possible. I would recommend using 'pi' as your default user, since there are no security risks associated with that (as long as you do basic hardening) and it will help avoid problems like this.

XECDesign, 2016-12-10:
Should be fixed by 172a20e008de1dc64b54cf4017b3e44d9dfcbdb8

closed 2016-12-10 by XECDesign