RPi-Distro/raspberrypi-ui-mods #10: pix-plym-splash disables splashscreen in prerm, even if it is not the current theme

mutability, 2017-01-17:
Couldn't find the source repository for pix-plym-splash, so putting this issue here as the next best option.

The pix-plym-splash prerm script removes the plymouth-related options from /boot/cmdline.txt unconditionally, even if the pix theme is not currently in use. This incorrectly disables plymouth if some other theme has been configured. It should not do that.

spl237, 2017-01-17:
The assumption is that if you are installing pix-plym-splash, you want the PIXEL splash screen - it isn't conditional on the PiX theme being selected.

I'll look to see if there is anything we can do to consider if the user has customised their splash screen first.

mutability, 2017-01-17:
pix-plym-splash can be installed without user intervention as part of an upgrade.

spl237, 2017-01-17:
Indeed it can - but the same upgrade also overrides user settings such as PiX theme for GTK and OpenBox, settings for the window manager and taskbar etc. We don't check to see whether the user has customised anything in these cases either - we just back up the user's settings. We could do that in this case as well, but the assumption is that if you are running PIXEL, you want your system to look like PIXEL.

mutability, 2017-01-17:
You should at least test that the current theme is pix before messing with cmdline.txt in prerm. If the current theme is not pix, then removing pix-plym-splash isn't going to change anything there anyway.

mutability, 2017-01-17:
If you can point me at the right repo for pix-plym-splash, I can send you a pull request to fix this.

spl237, 2017-01-20:
I've modified the prerm script to check for the pix theme being that currently selected - this will hopefully achieve what you need.

closed 2017-01-20 by spl237

mutability, 2017-01-20:
Thanks! Do you have a link to the commit?