RPi-Distro/raspberrypi-ui-mods #9: pix-plym-splash trashes existing plymouth theme selection

mutability, 2017-01-17:
Couldn't find the source repository for pix-plym-splash, so putting this issue here as the next best option.

pix-plym-splash unconditionally runs `plymouth-set-default-theme -R pix` in postinst.
This trashes any preexisting plymouth theme selection. It should not do that.

spl237, 2017-01-17:
As per my comment in your other issue - if you are installing pix-plym-splash, the assumption is that you want to select the PIXEL splash screen.

Can you suggest what we should check before running plymouth-set-default-theme? Even if this is still the default text theme, it doesn't prove that the user has not customised that theme themselves.

mutability, 2017-01-17:
As per my comment in my other issue - pix-plym-splash can be installed as collateral damage from a wider upgrade, since it's a dependency of raspberrypi-ui-mods.

Test for `[ $(plymouth-set-default-theme) = "pix" ]` is the obvious thing here (when run with no args, it tells you the current theme)

mutability, 2017-01-17:
Er, sorry, got my issues reversed here. Test for an uncustomized plymouth config file?

mutability, 2017-01-17:
Note that the specific case that I hit here was:

* Existing install based on pre-Pixel Jessie that has a third-party theme installed
* User does an apt-get dist-upgrade
* theme is replaced, user is confused

spl237, 2017-01-17:
As per other thread - the intention with an upgrade to PIXEL is that you get PIXEL, which includes the splash screen. This will be a one-off change on the upgrade to PIXEL - if a user has been able to customise their splash screen beforehand, they will surely be able to do so again?

mutability, 2017-01-17:
This is a customized image built from raspbian-ua-netinst; the average user does not know much about the internals; the new theme is installed as part of the image.

mutability, 2017-01-17:
Current images are omitting X entirely, FWIW, so raspberrypi-ui-mods making breaking changes won't be an issue going forward, but there are a few thousand existing installs already out there I need to manage.

mutability, 2017-01-17:
nb: this is how we handle it: https://github.com/flightaware/piaware-support/blob/6f73edbf879072a2b2b6c19851f661b9724a0d4e/debian/piaware-release.postinst#L49

piaware-release is the top level package that coordinates all the image customizations; if it's installed, this is definitely a piaware image. Even then, we only do the splashscreen setup once, on initial install (done when preparing the image) or on the first upgrade from a pre-splashscreen version.

spl237, 2017-01-20:
On thinking about this, I don't think there is a sensible way of detecting that changes have been made by the user - the behaviour of this package will remain, in that installing it overrides any current Plymouth theme.

closed 2017-01-20 by spl237

mutability, 2017-01-20:
Ok. Can you downgrade the dependency from raspberrypi-ui-mods? I am probably going to need to add a `Conflicts: pix-plym-splash` and it would be unfortunate if that made it impossible to install any of the UI customization.

spl237, 2017-01-20:
You said above you were moving to something not based on X anyway? In that case you don't need any of the UI customization, as it is all X based.

mutability, 2017-01-20:
The new image does not install X by default, right.

But this is something of a base image that does its one job, but is open to customization such as installing X and (ideally) raspberrypi-ui-mods.

At the moment I can't see how that is going to work - if the user installs raspberrypi-ui-mods then it will trash the plymouth config, and if I add a Conflict control line then that will prevent installation of raspberrypi-ui-mods. What do you suggest I do?

mutability, 2017-01-20:
Here are some of the cases I need to deal with (and how they currently break):

# Installation based on pre-PIXEL jessie

This has raspberrypi-ui-mods, piaware-release (pre-3.3.0), and no plymouth installed.

## Upgrade piaware-release first

Automated upgrade of piaware-release to 3.3.0 happens. 3.3.0 has a dependency on plymouth and plymouth-themes, they get installed. piaware-release postinst notices this is an upgrade from pre-3.3.0 and sets the plymouth default theme to piaware.

Automated dist-upgrade happens. The new raspberrypi-ui-mods pulls in pix-plym-splash. pix-plym-splash resets the plymouth default theme to pix.

End result: wrong theme is set.

### what if I add a Conflicts: pix-plym-splash to piaware-release?

dist-upgrade will probably (I haven't worked through the exact details) uninstall raspberrypi-ui-mods, which isn't great.

## dist-upgrade first

Automated dist-upgrade happens. The new raspberrypi-ui-mods pulls in plymouth, plymouth-themes, pix-plym-splash. pix-plym-splash resets the plymouth default theme to pix.

Automated upgrade of piaware-release to 3.3.0 happens. piaware-release postinst notices this is an upgrade from pre-3.3.0 and sets the plymouth default theme to piaware.

End result: correct theme is set

### what if I add a Conflicts: pix-plym-splash to piaware-release?

The piaware-release upgrade will probably fail, which is Really Bad.

# Installation based on post-PIXEL jessie

This is the "new image" that does not depend on raspberrypi-ui-mods, only piaware-release / plymouth / plymouth-themes are installed.

When building the image, the piaware-release postinst runs and sets the theme to piaware.

So far so good.

## user wants X

User wants X, so they install raspberrypi-ui-mods.
This depends on pix-plym-splash.
pix-plym-splash postinst resets the theme to pix.

End result: wrong theme is set.

## what if I add a Conflicts: pix-plym-splash?

Installing raspberrypi-ui-mods won't work.

# subsequent dist-upgrades

On any install that has pix-plym-splash, even if the theme is correctly configured, any dist-upgrade that updates pix-plym-splash will reset the theme to pix because its postinst does not do a check for the upgrade case.

End result: wrong theme is set.

This also means that giving users instructions on how to fix the theme after pix-plym-splash is first installed is futile, because it'll just get reset again unpredictably in the future

# more generally..

All of these automated updates act conservatively and set `DPkg::Options::=--force-confold` specifically to avoid unexpected changes to the user's config files during an automated operation. We really don't want to break local customizations that the user is made - it's not our hardware.

The raspberrypi-* packages and friends don't respect this and go and change config files anyway, which is pretty bad.

mutability, 2017-01-20:
Another option would be to ship a modified pix-plym-splash package in our repository (probably an empty package) that has a higher epoch version so that it gets installed in preference to anything that the raspberrypi.org repository provides, but that'd be a last resort since it'll actively break your stuff.