RPi-Distro/raspberrypi-ui-mods #14: Setting up raspberrypi-ui-mods often fails during pi-gen

malaire, 2017-09-01:
When running pi-gen, setting up `raspberrypi-ui-mods` often fails after messages:

    Setting up raspberrypi-ui-mods (1.20170725) ...
    update-alternatives: using /usr/bin/startlxde-pi to provide /usr/bin/x-session-manager (x-session-manager) in auto mode
    update-alternatives: using /usr/share/libgksu/debian/gconf-defaults.libgksu-sudo to provide /usr/share/gconf/defaults/10_libgksu (libgksu-gconf-defaults) in manual mode
    Backing up old config files...
    Backing up old config files...

Sometimes there is following warning after "Backing up old config files...", but often there is nothing else, build just hangs there indefinitely:

    (process:3536): dconf-WARNING **: failed to commit changes to dconf: Error spawning command line 'dbus-launch --autolaunch=6c0ae75983574faea5b1921152e7df64 --binary-syntax --close-stderr': Child process exited with code 1

Process tree when build hangs (build was started with command `time sudo ./build.sh |& tee mylog`, no warning message about dconf this time):

    sudo───build.sh───bash─┬─apt-get─┬─dpkg─┬─raspberrypi-ui-─┬─gsettings─┬─{dconf worker}
                           │         │      │                 │           └─{gsettings}
                           │         │      │                 └─{raspberrypi-ui-}
                           │         │      └─{dpkg}
                           │         └─{apt-get}
                           └─{bash}

I don't quite understand Debian configurations, but I think this might be happening during `gsettings set` lines in postinst which are about some screensaver settings.

Earlier issue at pi-gen: https://github.com/RPi-Distro/pi-gen/issues/102

malaire, 2017-09-01:
Better process tree - the problem occurs at very first `gsettings set` command in postinst.

    sudo ./build.sh
      └─build.sh -e ./build.sh
          └─bash /bin/bash
              ├─apt-get /usr/bin/apt-get install -y java-common oracle-java8-jdk libreoffice-sdbc-hsqldb sonic-pi python idle python3-pygame python-pygame python-tk python3 idle3 python3-tk python3-pgzero python-serial python3-serial python-picamera python3-picamera debian-reference-en dillo x2x raspberrypi-net-mods raspberrypi-ui-mods fonts-liberation2 smartsim python-pip python3-pip python3-numpy pypy minecraft-pi python-minecraftpi alacarte rc-gui sense-hat claws-mail tree scratch nuscratch greenfoot bluej nodered libgl1-mesa-dri libgles1-mesa libgles2-mesa xcompmgr geany piclone wiringpi pigpio python-pigpio python3-pigpio raspi-gpio python-gpiozero python3-gpiozero python3-rpi.gpio python-spidev python3-spidev python-twython python3-twython python-smbus python3-smbus python-flask python3-flask python-picraft python3-picraft python-sense-emu python3-sense-emu sense-emu-tools python-sense-emu-doc pprompt scratch2 python3-thonny
              │   ├─dpkg /usr/bin/dpkg --status-fd 17 --configure --pending
              │   │   ├─raspberrypi-ui- /bin/sh /var/lib/dpkg/info/raspberrypi-ui-mods.postinst configure 
              │   │   │   ├─gsettings /usr/bin/gsettings set org.gnome.desktop.screensaver lock-enabled false
              │   │   │   │   ├─{dconf worker}
              │   │   │   │   └─{gsettings}
              │   │   │   └─{raspberrypi-ui-}
              │   │   └─{dpkg}
              │   └─{apt-get}
              └─{bash}

XECDesign, 2017-11-20:
Those `gsettings` lines have been removed, since they were causing other problems too. When the next round of updates goes out, this shouldn't be a problem.

closed 2018-03-23 by XECDesign