RPi-Distro/raspberrypi-ui-mods #15: Often causes lxpanel to freeze when adding applications to the launch bar

zmarffy, 2018-01-14:
This is caused by rpi-ui-mods because if lxpanel is run by itself or run without ui-mods installed, it can add applications fine. Once it freezes on one app, that app can never be added back without causing a freeze again… without really messing around with settings.

spl237, 2018-01-14:
All ui-mods does to lxpanel is to change some of the initial settings; it shouldn't cause crashes, as it isn't changing any code. It is possible that something that you are adding to the launch bar is causing an unforeseen interaction with one of the new plugins.

Can you please provide a repeatable test case? Which application is it you are trying to add? Does the freeze happen every time or just occasionally? This isn't behaviour I have noticed myself, so I'll need more information to be able to repeat it if I am to debug it. Many thanks in advance.

zmarffy, 2018-01-14:
Occasionally: seems to happen often after an upgrade of ui-mods. Test case for me:

1. Right click panel
2. Click Application Launch Bar
3. Click Preferences
4. Double click Accessories
5. Click Terminal
6. Click Add
7. lxpanel freezes until restart and does not add Terminal

I have fixed this when it happens by deleting the entire lxpanel config and remaking it myself with the GUI.

spl237, 2018-01-14:
OK, I've just tried that several times here, and it works fine for me. I suspect something in your panel config is causing a problem. Could you please cut and paste the contents of ~/.config/lxpanel/LXDE-pi/panels/panel on here the next time you see a crash? I can then investigate.

If you wanted to do some more narrowing down, you could try removing the plugins from the panel one at a time and see if any of them make a difference.

zmarffy, 2018-01-19:
Will do so next time it crashes.

XECDesign, 2018-03-23:
Inactive. Please re-open with the additional information if this is still an issue.

closed 2018-03-23 by XECDesign

zmarffy, 2020-11-25:
Back when I was a wee lad who knew barely anything about computers. But I can say it never happened again… but maybe that's because I no longer use the GUI.