RPi-Distro/raspberrypi-ui-mods #20: lxappearance exits with "Segmentation Fault"

COOLIRON2311, 2018-12-28:
However the issue can be fixed by moving `/usr/share/themes/PiX`.

spl237, 2018-12-28:
I'm aware of the obconf issue, and will be fixing that when I am back in the New Year, but I can't repeat the lxappearance problem - that runs fine for me with no segfaults on my latest image.

Is there anything in particular I need to do to trigger the crash, or does it just happen as soon as you run the application?

COOLIRON2311, 2018-12-28:
It happens instantaneously. I think the reason for that is lxappearance-obconf package.

spl237, 2018-12-28:
OK, that would make sense - it is obconf which is crashing, in that case, not lxappearance. 

That's a known fault - I'll fix it in a week or so. For what it's worth, we don't recommend using either lxappearance or obconf to configure the system any more - both have been replaced with pipanel, but that doesn't configure as many aspects of the system as the two previous applications.

closed 2020-04-09 by spl237