RPi-Distro/raspberrypi-ui-mods #22: Adding Desktop Pager removes ability to switch/move/resize/close windows

scruss, 2019-11-18:
(was [Adding Desktop Pager removes ability to switch/move/resize/close windows · Issue #1336 · raspberrypi/documentation](https://github.com/raspberrypi/documentation/issues/1336), but noted it was in the wrong place.

**Actions in desktop**:

1. right click in the menu bar, “Add/Remove panel items”

2. in Panel Preferences click “Add”, then select “Desktop Pager” and “Add”. 

3. Right-click on newly-created one-page pager,  and choose “Desktop Pager Settings”. Add a number (> 1)  in the Desktops tab.

4. Close settings.

**Expected result**: use desktop as normal, but with multiple workspaces.

**Actual result**: multiple workspaces are enabled, but the ability to switch/move/resize/close windows is lost. The only way to switch/manipulate windows is through right clicking the window indicator in the top bar.

This may be an issue with obconf writing unexpected values into the user's desktop configuration. Seems to be a recent issue, perhaps since Stretch. Didn't appear to be an issue around 2015. Not a stock/upstream LXDE user issue, either: I'm a long-time Lubuntu user and I can add many workspaces without problems. Seems also to be a problem on Raspberry Pi Desktop (for PC and Mac) too.

forum reference: https://www.raspberrypi.org/forums/viewtopic.php?f=116&t=251268

**Update**: as noted by user petr2 [here](https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=251268&p=1568003#p1567525), creating workspaces via `obconf` does not cause this problem to appear.

I use multiple workspaces because I have attention deficit/sensory problems, and one workspace per task allows me to focus.

spl237, 2019-11-18:
I've modified obconf, which is the tool which is called by the desktop pager plugin to set up desktops, so that it loads the default configuration correctly from global openbox config files; this was a change made to openbox for the buster release, and it is what causes the creation of a new desktop in the tool to change the window frame configuration.

I don't know of any way of creating workspaces other than with obconf - as above, right-clicking the desktop pager and choosing Desktop Pager Settings opens obconf anyway.

I have not been able to repeat the problem with it not being possible to drag or close windows; both work fine for me with multiple desktops.

The changes to obconf will be available from apt within the next couple of hours - sudo apt update / sudo apt dist-upgrade will get them.

scruss, 2019-11-18:
Thank you! The issue seems resolved with your `obconf-1:2.0.4+git20150213-2+rpt2`. One needs to enable/set workspaces in obconf and then add the Desktop Pager plugin. Previously, going direct to the adding the Desktop pager plugin called up obconf — but in the past, also caused the window locking problem. The theme is also retained now.

(Could do without `lxterminal.conf` being blown away in desktop update, though: past age 50, anything smaller than custom screen/terminal font sizes might as well be in 3 pt Flyspeck Squinty … but that's another issue)

spl237, 2019-11-19:
> (Could do without lxterminal.conf being blown away in desktop update, though: past age 50, anything smaller than custom screen/terminal font sizes might as well be in 3 pt Flyspeck Squinty … but that's another issue)

We don't anything to lxterminal.conf on updates any more. Nowadays, configuration files will only be changed if the user presses one of the "defaults" buttons in the Appearance Settings dialog.

closed 2019-11-19 by spl237

scruss, 2020-04-09:
This issue seems to be back. It looks like there may be something not updating a theme setting. The instant one changes the number of desktops in obconf, the theme seems to change to Clearlooks (blue borders) from PiXflat (grey borders). It's here that the window move and control widgets cease working. Checking the theme tab in obconf makes the theme switch to Artwiz-boxed (dark grey borders), the first theme in the list.

I can repeat this every time. Just tried it on two different machines with identical results.

Installed packages/versions:

    gtk2-engines-pixflat:armhf              	0.3
    lxappearance-obconf                     	0.2.3-1+rpt1
    lxpanel                                 	0.10.0-2+rpt7
    obconf                                  	1:2.0.4+git20150213-2+rpt2
    pixflat-icons                           	0.2
    raspberrypi-ui-mods                     	1.20200218

## Steps to reproduce

1. Right click on Task Bar

2. Select “Add / Remove Panel Items”

3. In *Panel Preferences / Panel Applets* tab, select “Add”

4. In *Add plugin to panel* window, select “Desktop Pager” and click “Add”

5. Close *Panel Preferences* window

6. Right click in Desktop Pager panel widget, and select “Desktop Pager Settings”

7. In *Openbox Configuration Manager / Desktops* tab, reduce the number of desktops using the -/+ buttons. On doing this, the window border colour will change from grey to blue as if a theme has changed (to Clearlooks?) and window move by dragging top frame and control icons will become inactive.

## Steps to work around/repair

1. Click on Raspberry menu

2. Go to *Preferences* → *Appearance Settings*

3. In the *Appearance Settings / Defaults* tab, select a suitable screen size and click *Set Defaults*. Window move and control icons will become active again

4. Screen will reset to default RPF background and font. If necessary, change these to your preferred ones

5. Close *Appearance Settings* window

6. Right click in Desktop Pager panel widget, and select “Desktop Pager Settings”

7. In *Openbox Configuration Manager / Desktops* tab, set the number of desktops using the -/+ buttons

8. Close the *Openbox Configuration Manager* window. 

Number of desktops and control widgets remain stable through further changes.

spl237, 2020-04-09:
Sorry, but I can't repeat this. We've not made any changes to openbox or obconf since this was fixed previously; obconf is now correctly reading either the local or the global config file depending on which exist, and writing to the local file. If I have no local configuration file for openbox - i.e. the global system file is used - I do not get a theme change when adding a desktop; I also do not get a theme change if there is already a local config file and I add a desktop.

In your report above, you list in the steps to repeat  "7. In Openbox Configuration Manager / Desktops tab, _reduce_ the number of desktops using the -/+ buttons..."

I don't understand this instruction, as the default configuration in a clean image is to have a single desktop; it is not possible to reduce the number of desktops from 1. So something in your initial configuration is different from the default - this may be what is causing the problem for you.

scruss, 2020-04-09:
When you open the Desktops tab, it offers 4 desktops from a fresh install (with desktop pager added). Only if you change this number with the -/+ buttons does the actual number of desktops change. And that's when the issue starts. If you just click 'close' from the Desktops tab without changing anything, it won't change the number of desktops.

The problem may not be in obconf: it may be in how the default theme is set up. I don't know the internals. I've tried to give the best analysis I can. The 'freezing' issue is still being reported on the forums: here's one from last month - [Close, Minimize and full screen buttons not responding. - Raspberry Pi Forums](https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=268497&p=1637207#p1629668)

spl237, 2020-04-09:
On a new image, the default number of desktops is defined in the system openbox config XML file; this is /etc/xdg/openbox/lxde-pi-rc.xml. The default version of this file contains the line <number>1</number> in the section <desktops> - this is what determines the default number of desktops. 

I have just double-checked this - on a clean flash of the February 2020 image, launching obconf from the terminal with an strace running, this is the file which is accessed, and obconf offers me 1 desktop as the initial value.

I'm sorry, but something is wrong somewhere if your system is defaulting to 4 desktops. The default Pi theme for openbox definitely only offers a single desktop by default, and I cannot repeat the problem you describe above on a clean flash of the latest image. Are you definitely using the most recent image - dated 13 February 2020?

scruss, 2020-04-21:
I can't repeat this from an absolutely fresh Feb 2020 installation. Since the system I'm using is a Raspberry Pi 4, it can't be *that* old an image, and I haven't edited any of the config files outside using obconf.

spl237, 2020-04-21:
If it's not reproducible in the current image, then it's not something we'll do anything about. If you can reproduce it in a future image, feel free to mention it here.

khaoskosmos, 2021-08-14:
I was able to reproduce it. The window controls freeze up just as described here, after using openbox configuration manager.

pm0u, 2023-05-11:
I am able to reproduce this on a fresh SD install.

64 bit 2023-05-03, Pi 4b 4G & 8G, Raspberry Pi touch Display.

To reproduce:
- flash SD
- Boot Pi, complete all necessary configs
- Add the desktop pager to the menu bar
- Menu bar breaks and is hidden on half the width of the screen.
- Move the pager to a visible part of the menu bar
- Click to another space
- Screen background turns brown, all windows are unresponsive.

spl237, 2023-05-11:
We're not using openbox on newer Pis any more; we have switched over to mutter, so obconf won't do anything.

We stopped supporting multiple desktops a long time ago - we're doing nothing to break it in the upstream code, but it has probably bit-rotted in there by now. We've never supported the desktop pager plugin, which is also likely to have bit-rotted.

This is a wontfix, I'm afraid