RPi-Distro/raspberrypi-ui-mods #25: Localisation issue with the custom lxpanel profile config file

emanueleg, 2020-06-09:
Hardcoded names for both the ["Run..."](https://github.com/RPi-Distro/raspberrypi-ui-mods/blob/02199607969e9fdcfa38dac422264351583b3f36/etc/xdg/lxpanel/LXDE-pi/panels/panel#L38) and ["Shutdown..."](https://github.com/RPi-Distro/raspberrypi-ui-mods/blob/02199607969e9fdcfa38dac422264351583b3f36/etc/xdg/lxpanel/LXDE-pi/panels/panel#L45) menu items should be removed from the custom lxpanel profile config file.

Instead, the default "Run" and "Logout" lxpanel menu entries (see lxpanel-buster [`plugins/menu.c`](https://github.com/raspberrypi-ui/lxpanel-buster/blob/77133b14b7a18044d0b98b13963c99ea44c15df8/plugins/menu.c#L733)) should be modified/localised through the lxpanel po file.

XECDesign, 2020-06-11:
@spl237

spl237, 2020-06-11:
Fixed - many thanks.

closed 2020-06-11 by spl237