RPi-Distro/raspberrypi-ui-mods #26: Install or upgrade hangs until GUI alert is acknowledged

gblandst, 2020-06-30:
When upgrading or installing raspberrypi-ui-mods in a virtual console, it may display the following message:

`The desktop has been updated.
To apply the updates, please reboot your Pi, and then select one of the options on the Defaults page in Appearance Settings.`

Then the install script apparently hangs.  There appears to be no way for the user to acknowledge this and continue; seemingly the only option is to interrupt the process, leaving dpkg in a bad state.

The reason for the hang is not obvious unless we switch to the GUI session, and see that an alert has popped up, and that the install script is waiting for the user to dismiss this alert before continuing.

This is not expected behavior from an apt upgrade: most of the interaction happens in the console/terminal session. 
 Furthermore, this alert doesn't do anything except inform the user, which means there's no real need to interrupt everything (including the rest of what may be a lengthy system upgrade) until the alert is acknowledged.  It's a frustrating user experience when using a virtual console.

An ideal fix would be to not fire a GUI alert when running from a virtual console.  Regardless, it would also be good to make any such information-only GUI alert fire-and-forget, so that if the user didn't already see the message in the console output, it would be waiting for them in the GUI whenever they return there, and any subsequent upgrades and commands can continue to run.  If none of that is acceptable and it's essential to halt everything so that the user acknowledges this message, the user should have the option to do so in the same console where the install/upgrade is running (e.g. by pressing a key), rather than requiring the user to acknowledge it out of band, assuming they even know where to look for it.

spl237, 2020-07-01:
The message is required to inform the usert that they need to take action if they want the changes from the update. It also automatically clears after being displayed for 30 seconds. The current behaviour will not be changed.

closed 2020-07-01 by spl237

gblandst, 2020-07-01:
The dialog disappears after 30 seconds when run in a terminal in the GUI.

This behavior does not occur when running apt in the virtual console; there, it hangs indefinitely.