RPi-Distro/raspberrypi-ui-mods #36: mutter should not be a hard dependency

maxnet, 2021-12-11:
mutter not only causes problems with realvnc -for which you have a test in your startup script-, but also other VNC server stuff like x11vnc.

The easiest workaround for someone wanting to run that would be "sudo apt remove mutter"
But right now that also uninstalls raspberrypi-ui-mods because you list mutter as depends in debian/control.
I don't see any hard requirement on mutter in this package since you test for the existence of /usr/bin/mutter in your script though, so can mutter be made a Recommends instead?

spl237, 2021-12-11:
I don't really think that is a valid long term solution, unfortunately. We will be phasing out openbox completely as we move forward, and mutter will be the only option as a window manager. RealVNC are already working on improving the compatibility of their software with mutter; the authors of other VNC servers will need to do the same, because most desktops are going to move away from X to solutions using compositing window managers.

The easiest way to disable mutter isn't actually to uninstall it, but to edit /usr/bin/startlxde-pi - there is a chunk at the bottom of this file which checks to see how much memory is available and selects either openbox or mutter accordingly. If you change the value this tests against from 2048 to something bigger than the RAM of any Pi - 12048 works fine - then you will get openbox instead of mutter all the time. But as above - this should be regarded as a temporary workaround rather than a long-term solution.

closed 2021-12-11 by spl237

maxnet, 2021-12-11:
>The easiest way to disable mutter isn't actually to uninstall it, but to edit /usr/bin/startlxde-pi 

I am not a big fan of editing files, as that usually does not survives package upgrades...

>RealVNC are already working on improving the compatibility of their software with mutter; the authors of other VNC servers 
>will need to do the same, because most desktops are going to move away from X to solutions using compositing window 
>managers.

I am not sure if this is actually something authors of VNC server can fix.

May be a VC4 specific problem, rather than something of compositing window managers in generally.
Have the impression that reading back pixel data after the GPU did its thing, is noticeably slower than on other devices.
Noticed it behaves faster if I tell it to make a copy of the entire 1920x1080 framebuffer before doing anything (x11vnc has a "-snapfb" option that essentially does that), instead of reading multiple smaller regions which it normally does.
Like you are storing things in a tiled format, that does not allow accessing small regions in an efficient matter.