RPi-Distro/raspberrypi-ui-mods #37: [Question / Feature request] Alt+Tab behavior with Mutter

Caleb9, 2022-01-23:
Hello,

I've been directed over here with my issue from [StackExchange question](https://raspberrypi.stackexchange.com/q/133938/141734) and [Mutter GitHub](https://gitlab.gnome.org/GNOME/mutter/-/issues/2113).

In Raspberry Pi OS Bullseye, which uses Mutter as default Windows Manager, I noticed that Alt+Tab key combination switches only between current and previously focused windows, i.e. when I have 3 windows open in order A, B, C then I can switch only between B and C, while A stays in the background. So when I need to switch to window A, I need to grab a mouse to change focus.

Is there a way to configure this behavior somehow to be able to switch through all open windows, in a more "classical" way? If so, how can I do that, otherwise is there a chance such configuration could be added or the default behavior changed?

Please note that an "application switcher" of any sort (usually in a form visual overview of all open windows when pressing Alt+Tab and keeping Alt pressed) is completely secondary for me, personally. For productivity reasons I just prefer to be able to switch through more than 2 most recent windows with Alt+Tab.

The StackExchange question mentioned above proposes a workaround: switching Window Manager to Openbox, which seems to have worse performance than Mutter :(.

Thanks in advance.

Drdrm46, 2022-01-23:
In GNOME, Alt Tab works properly with visual overview icons of open applications and selection of any - to achieve this gnome-shell augments the limited capability of mutter. LXDE (and LXDE-pi) were designed to work with openbox, not mutter so the full Alt Tab capability seems to have fallen through the cracks. This is a raspberrypi-ui-mods issue because there is no reason to say there is an issue with mutter or LXDE. To recover from this regression LXDE-pi would have to be enhanced to work correctly with mutter. I think this a major issue because it makes Raspi OS on Pi 4 more difficult to use for the committed desktop user. All other major Linux desktops (and previous versions of Ras Pi OS) implement Alt Tab properly with selection from visual overview icons of open applications. Raspi OS has thus become an outlier in this regard

spl237, 2022-01-23:
I believe this is a bug in mutter itself. If you do shift-alt-tab, which should be the same operation but switching through windows in the reverse order, it correctly steps through all windows, not just the last two. The keybindings are correct, and having had a quick look at the mutter code, the same function is being called for both operations, but it is failing in the forward case.

For the time being, shift-alt-tab is a workaround; I'll see if I can work out why the forward case doesn't work in mutter and patch it.

Drdrm46, 2022-01-23:
In GNOME, if you press Alt Tab and release Alt quickly (within half a second) you switch to the last active application. Then if you do the same thing you switch back to the original, ie quick Alt Tabs toggle between the last two open applications just like in RPi OS with mutter. 
However, in GNOME, if you press Alt Tab and hold Alt down for more than half to 1 second, a row of icons for open applications appears and you can tab to whichever application you want. In RPi OS with mutter, this does not happen however long you wait. In the wikipedia article on Window Managers for X it claims that the  selection icon functionality is provided by gnome-shell, augmenting the functionality of mutter. I must say I think the icons are very important for ease of selecting applications by keyboard - a function which almost all users of LInux and Windows desktops must know about.

Drdrm46, 2022-01-24:
Thanks for quick response. As GNOME implements Alt Tab cycling fully through code which must be outside mutter but works with mutter, there must be some code that can be lifted from gnome-shell plug-in to provide tab cycling. - and the GNOME team are unlikely to accept that mutter itself needs to be fixed in this regard. I hope that a practicable solution can be found in due course. Thanks again.

Caleb9, 2022-01-24:
Maybe this is obvious by now, but just for completeness of my initial post here: pressing Alt+Tab and releasing Alt switches to previously focused window. The window we had focused before pressing Alt+Tab then becomes "previously focused window" - so this works on Raspberry Pi OS with Mutter as expected.

The difference with how it usually works elsewhere is when you keep Alt pressed and then repeatedly press Tab. I would expect in such situation to be able to loop through all the open windows, but on Raspberry Pi OS it only switches between last two windows.

Really appreciate you guys looking into it, thanks! :)

Jaciss, 2022-02-03:
I have followed @Caleb9's posts around the interwebs and landed here. I cannot leave without a second to this: **please fix alt+tab to work as users expect it to**. For something like a pi, which is more likely to have use cases where users do not have access to a mouse it is rather vital. I almost never use a mouse and have used alt+tab for window navigation since operating systems *had* windows. It is ubiquitous.

Obviously I will keep looking into this and report back anything useful in case any other WIMP haters come here looking for a better solution than alt+shift+tab. Thank you all for your time.

spl237, 2022-02-03:
I've fixed this issue. It turns out that mutter didn't actually have any window cycling code at all; just the application switching which (as described above) is for a different purpose. The code did distinguish between application switching and window cycling, but both mapped onto the application switching code, so I wrote code to cycle between windows and added it. I think it should do what people are requesting above.

I'm working on a few more small fixes for mutter issues - when they are all fixed and tested, we will release a new version which will include window cycling; hopefully within the next week or so. But I promise - a fix exists, and is on the way...!

Drdrm46, 2022-02-03:
Thanks and I am hoping we can have full Alt-Tab capability i.e. not just cycling alone as normally provided by Alt-Esc, but cycling with selection icons and/or snapshot popups as in LXQt+Lite which works well on the Raspberry Pi 4.

spl237, 2022-02-03:
> Thanks and I am hoping we can have full Alt-Tab capability i.e. not just cycling alone as normally provided by Alt-Esc, but cycling with selection icons and/or snapshot popups as in LXQt+Lite which works well on the Raspberry Pi 4.

No, I'm afraid that won't be possible - just cycling without pop-ups. Implementing a selection dialog with icons is a lot more work...

Jaciss, 2022-02-04:
@spl237 you are officially my hero of the day and named thus in my gratitude journal. You will probably save a lot of geeks significant cumulative time and frustration using their Pi with this. On behalf of myself and those who will probably never even know: Thank you!!! Switching back to openbox was not an ideal solution. Obviously ;)

On the cycling menu: yes, a number of people would probably like this but it's a "nice to have" and not a "broken out of the box" so thumbs up. People can get to everything with just a keyboard: win.

spl237, 2022-02-04:
The new version of mutter has now been uploaded to apt, and is syncing to the servers as I type; it should be available later this afternoon. "sudo apt update" followed by "sudo apt full-upgrade" will install it.

Key mappings are that alt-tab / shift-alt-tab cycles through all open windows; ctrl-tab / shift-ctrl-tab cycles through all open windows in the currently-focussed application, and alt-esc / shift-alt-esc now switches between applications in forward and reverse chronological order; this is the functionality which was on alt-tab and shift-alt-tab.

closed 2022-02-04 by spl237

Drdrm46, 2022-02-04:
Thanks. That's so  much better. Good stuff.

Caleb9, 2022-02-04:
I just updated and it works - fantastic, thanks @spl237.

Not complaining, since I can achieve everything I need, but just to note that the current behavior is still slightly different than most commonly experienced, where pressing alt-tab switches to previously focused window and _pushes the window we're tabbing out of to the top of the stack_ when alt is released. So e.g. in Cinnamon or Windows alt-tab combines behavior of current Mutter's alt-tab and alt-esc, if that makes sense.

Maybe an idea for future improvement, but otherwise I think it's perfectly fine to accept that there's simply a different way to achieve both things (alt-tab and alt-esc) which was not possible previously. Thanks again. :)

Drdrm46, 2022-02-05:
Alt Tab behaviour is different from expected – different from behaviour with openbox.

The expected behaviour is that Alt Tab does ‘toggling’ and ‘cycling’ – not just toggling as was the case with mutter Now we have reverse cycling but not toggling.

Standard behaviour (Openbox) – Toggling and Cycling
Open A B and C so C has focus
-	Alt Down Tab Down  Tab Up Alt Up  changes focus from C to B. A further Alt Down Tab Down  Tab Up Alt Up changes focus back to C. This allows quick toggling between C and B without A coming into focus (Toggling).
-	Alt Down Tab Down Tab Up Tab Down Tab Up Alt Up moves focus from C to A via B (Cycling). Subsequently one can toggle between C and A via successive  uses of Alt Down Tab Down  Tab Up Alt Up
The above is also what happens in other distros such as GNOME and LXQt. and is what most people will be familiar with.

With new mutter modification – No toggling, inverse cycling
-	Alt Down Tab Down  Tab Up Alt Up changes focus from C to A.  A further Alt Down Tab Down  Tab Up Alt Up changes focus to B. Alt Tab does not provide the quick toggle capability. (No Toggling but a sort of inverse cycling.)
-	Alt Down Tab Down Tab Up Tab Down Tab Up Alt Up moves focus from C to B via A (Inverse cycling)

Alt Shift Tab works the other way round in both cases 
In newly modified Mutter,
-	Alt Shift Tab behaves as Alt Tab in Openbox
-	Alt Tab behaves as Alt Shift Tab in Openbox.

I know Alt Esc behaves like Alt Tab used to with mutter but Alt Esc is barely known except by specialist IT people.

The ‘normal’ familiar way of using Alt and Tab keys which is easiest for the hand  is the openbox method as Alt Shift Tab arguably requires two hands. In fact most people will only know about and use Alt Tab. Both Alt Esc and Alt Shift Tab are probably very rarely used except by IT specialists. Speaking personally I have not found reason to use these alternatives any time in the last 30 years since windows 3.1 was introduced.

I am hoping we can get back to ‘normal’ behaviour at some point in the not too distant future.

This is very complicated and I hope I have got the details right – but the big picture is that Alt Tab works as expected with Openbox and it would be nice (or perhaps highly desirable ?) if the behaviour were the same as when using mutter (albeit without the pop-ups).

spl237, 2022-02-05:
> Alt Shift Tab works the other way round in both cases In newly modified Mutter,
> 
> * Alt Shift Tab behaves as Alt Tab in Openbox
> * Alt Tab behaves as Alt Shift Tab in Openbox.

In that case, you can fix it easily yourself - just install dconf-editor and swap the keybindings - they are under either /org/gnome/mutter/keybindings or /org/gnome/desktop/wm/keybindings

spl237, 2022-02-05:
Actually - don't worry about changing keybindings; I've now inverted the sense of the forward and reverse directions for cycling in the mutter code - there should be a new version which does things the other way around in apt in the next day or so.

Drdrm46, 2022-02-05:
I tried changing the wm schema successfully and have now changed back. Please check and double check very, very carefully that operation after change is exactly like with openbox, if that is what you are attempting to achieve, as I am not too sure exactly how things work and beginning to suspect I might have may have made an erroneous assumption - I would have to a lot more experiments to check whether what I have been saying is absolutely right

spl237, 2022-02-05:
In reality, openbox and mutter are not really doing the same thing anyway; I think openbox maintains a list of last accessed windows which is only updated when alt is released; I am using the stacked order of windows in mutter as a proxy for that, which will give similar behaviour in most circumstances, but is not quite the same thing.

As users can change the schema to select whichever direction works best for them anyway, I'm inclined to just accept your statement that the default sense of cycling was inverted compared to openbox, and have made the change accordingly.

Drdrm46, 2022-02-05:
Ok  this seems to be as near to openbox as can be reasonably achieved - symmetries and asymmetries between the directions are confusing - I must have been half asleep last night.  Still don't have the toggling on Alt Tab in  fact whether or not one releases the Alt key between successive tabs does not seem to make any difference. Still need to use Alt Esc for toggling. So much better than before though. Thanks from an enthusiastic proponent of Raspberry Pi as a desktop replacement who has been using Raspberry Pi's as sole desktop PC for self and family since Jessie introduced on Raspberry Pi 2.

Jaciss, 2022-02-05:
I woke up wanting to make a demo, though it seems I'm a bit late. I don't have time to play more right now but I will be back.

```
#!/bin/bash

# A *demonstration* of Alt+Tab behaviour on Raspbery Pi Bullseye (Mutter)
# for xprop: `sudo apt install x11-utils`
# for xdotool: `sudo apt install xdotool`
# 
# You can bind this to Alt+Tab with xbindkeys
# something like this (assumes file named alttab on $PATH):
# ```
# sudo apt install xbindkeys
# xbindkeys --defaults > ~/.xbindkeysrc #This will overwrite ~/.xbindkeysrc!
# echo '"lxterminal --command=alttab"' >> ~/.xbindkeysrc
# echo '  Alt + Tab' >> ~/.xbindkeysrc
# echo "\"lxterminal --command='alttab 1'\"" >> ~/.xbindkeysrc
# echo '  Alt+Tab+Release' >> ~/.xbindkeysrc
# xbindkeys --poll-rc #[re]load rc file
# ```

# OR

# use it as originally intended - run the script (equivalent of pressing alt+tab) 
# and hit the tab key n times to test out the behaviour

use_xbindkey_hack=0 #this isn't designed for real-world use; you have been warned

# count tab presses
if [[ $use_xbindkey_hack -eq 1 ]]; then
	ttfile='/tmp/tabtimes.txt'
	tnow=`date +%s`
	tabtimes=0
	lasttime=0
	if [ -f "$ttfile" ]; then
		read tabtimes lasttime < $ttfile
	fi
	tdiff=$(($tnow-$lasttime))
	if [[ $tdiff -lt 3 ]]; then
		((tabtimes+=1))
	elif [[ $tdiff -ge 3 ]]; then
		tabtimes=1
	fi
	lasttime=`date +%s`
	echo "$tabtimes $lasttime" > $ttfile

	# if not told to act (i.e., alt not yet released) stop at counting
	if [ $# -eq 0 ]; then
		exit
	fi
	num_presses=$(($tabtimes+2))
else
	i=2
	num_presses=2 #skip "panel" window and active window (+2)
	while [ $i -ge 0 ]
	do
		IFS= read -t 1 -n 1 input
		# is it a tab?
		if [[ $input = $'\t'* ]]; then
		   ((num_presses+=1))
		   ((i+=1))
		# timeout if no input
		elif [[ -z "$input" ]]; then
		   i=0
		fi
		((i-=1))
	done
fi

# echo && echo "Tab presses: $num_presses"

# get window stack list
stacklist=`xprop -root | grep '_NET_CLIENT_LIST_STACKING(WINDOW)' | sed 's/.*#//'`

# reverse the order
stacklist=`echo $stacklist | sed 's/, /\n/g' | tac | sed ':a;$!{N;ba};s/\n/, /g'`

# loop through stack until we get to our $num_presses index; raise window
j=0

echo $stacklist | sed -n 1'p' | tr ',' '\n' | while read win_id; do
    ((j+=1))
#   wname=`xprop -id $win_id WM_NAME` #you can get _NET_WM_ICON as well
#   echo "$j: $win_id $wname"
    if (( $j == $num_presses )); then
# 	echo "raise $win_id"
	xdotool windowactivate $win_id
    fi
done
```

Drdrm46, 2022-02-05:
Thanks  I've been planning to try xbindkeys, not least because I want to add custom keybindings and I have not succeeded using dconf-editor and gsettings even though my configuration seems OK. Looking forward to when you come back with more !

Jaciss, 2022-02-05:
> Thanks I've been planning to try xbindkeys, not least because I want to add custom keybindings and I have not succeeded using dconf-editor and gsettings even though my configuration seems OK. Looking forward to when you come back with more !

I don't much like xbindkeys which is probably why the binding only works properly half the time. I have to figure out what causes presses and when then tweak the script for that so I can polish the timing. xbindkeys is the real deal, it just makes using other machines w/out personalized bindings/etc annoying so I avoid it. Of note: **xbindkeys is picky about spaces**. Both in indentation and (previously, maybe still) in the actual definitions so you might check that.

Personally, I rarely bother with conf files like that. Stuff changes and breaks a lot but things like xbindkeys have been around and working forever and it's easy to back up with your other home settings so you don't have to reconfigure so much on new machines of whatever flavour. Ditto for small shell scripts ;)

On the bright side, this little shell script's logic at least works flawlessly, and the point was to show how alt+tab works and how to access and use the window stack. It pulls up the expected window every time for n presses...just the xbindkeys bit doesn't always interpret what it gets the way it should and sends the wrong number of presses. Fixable but maybe not worth the time for a short-term hack. I do hope it's useful to people, regardless.

Drdrm46, 2022-02-06:
Thanks for your script - it will help me when I learn xbindkeys and xdotool.
 It seems that we are tantalisingly close to standard alt tab operation - 'just' needing to swap the top two entries on the windows stack at the right time - but I understand this is not easy.

Jaciss, 2022-02-06:
> Thanks for your script - it will help me when I learn xbindkeys and xdotool. It seems that we are tantalisingly close to standard alt tab operation - 'just' needing to swap the top two entries on the windows stack at the right time - but I understand this is not easy.

xbindkeys is pretty straightforward for simple things - just quote your command on the first line and  `xbindkeys --multikey` or --key will show you what to enter for the second line's key definitions. The two lines in my script to install it and its example rc file are probably all you need to get started.

xdotool is one of my few manditory scripts. Boot-->`lxterminal --command=script_using_xdotool`-->open everything you usually open and type everything you usually type. Instant workspace. If you're a bit careful (like use `xdg-open` for browser stuff) your automaton will work on most machines. Forevars.

I appreciate your thanks. I wish I could get us closer. I **updated my script and it works reliably now**, with the caveat that you must release Alt first. Ctrl+Tab and Ctrl+Shift+Tab don't work in lxterminal or chromium anymore, and natural scrolling config still doesn't work, so...not using Mutter much right now.

spl237, 2022-02-06:
>Ctrl+Tab and Ctrl+Shift+Tab don't work in lxterminal or chromium anymore, and natural scrolling config still doesn't work, so...not using Mutter much right now.

If those being used in mutter is causing you a problem with their use in applications, just remove the bindings from mutter. Install dconf-editor (sudo apt install dconf-editor) and in /org/gnome/desktop/wm/keybindings, delete the association for cycle-group and cycle-group-backward - mutter will then not grab those key combinations and they will be passed straight to apps as before.

Jaciss, 2022-02-06:
You are again a hero, @spl237 - while neither myself nor my install scripts will be happy with the now very pretty dconf-editor your tip helped me discover that they're moving things into central binary files, which is why config files are failing. You can still do everything via the command line. This solved my issues:

```
gsettings set org.gnome.desktop.peripherals.touchpad natural-scroll true
gsettings set org.gnome.desktop.wm.keybindings cycle-group-backward []
gsettings set org.gnome.desktop.wm.keybindings cycle-group []
```

And gsettings will help you out with tab-completion, even. Beautiful.

Jaciss, 2022-02-07:
So I decided I'd stick with Mutter full-time after all. Which means alt+tab must work as I'm used to. Which means I updated my script again and it now functions just like alt+tab on pretty much every other system out there. Excepting the GUI menu showing you a window selection list. Happy alt+tabbing!

Drdrm46, 2022-02-07:
OK so I've started using mutter with xbindkeys custom shortcuts (as gsettings using following does not seem to work presumably because capability is not implemented in Raspberry Pi OS) and posted on Raspberry Pi forum.

#!/bin/bash
gsettings set org.gnome.settings-daemon.plugins.media-keys custom-keybindings "['/org/gnome/settings-daemon/plugins/media-keys/custom-keybindings/custom0']"
gsettings set org.gnome.settings-daemon.plugins.media-keys.custom-keybinding:/org/gnome/settings-daemon/plugins/media-keys/custom-keybindings/custom0/ name "Emergency Power Off"
gsettings set org.gnome.settings-daemon.plugins.media-keys.custom-keybinding:/org/gnome/settings-daemon/plugins/media-keys/custom-keybindings/custom0/ command "/usr/sbin/poweroff"
gsettings set org.gnome.settings-daemon.plugins.media-keys.custom-keybinding:/org/gnome/settings-daemon/plugins/media-keys/custom-keybindings/custom0/ binding "<Alt>q"

I still have my family on openbox and would be grateful of an updated copy of your script !

Thanks to all and all best

Jaciss, 2022-02-07:
@Drdrm46 
> I still have my family on openbox and would be grateful of an updated copy of your script !

I've just been updating the [original bash alt+tab post](https://github.com/RPi-Distro/raspberrypi-ui-mods/issues/37#issuecomment-1030676934) to avoid people getting old versions. Without the GUI showing what you're selecting I don't know that it'd be ideal for casual users. As in definitely not. It does make Mutter usable for people who have used alt+tab too long to change ;)

>  angle brackets is not showing in the preview or the post for some reason.

Use can single backticks to enclose single-line code snippets, and triple backticks for multi-line. If you type the language name after the triple backticks markdown-based systems will usually highlight the code. Like so:
```
```bash
echo "Hello, world!"
```

Drdrm46, 2022-03-08:
It is still desirable to have not only Alt Tab 'Select and Toggle' capability as provided by Jaciss's excellent script, **but also** a popup splash showing a list of active windows for selection when using Mutter windows manager.

I wrote a popup daemon in python 3 to work together with a modified version of Jaciss's Alt Tab script - modified to allow time for selecting the required window from the popup. The script is a drop-in replacement for Jaciss's script and communicates with the popup daemon through temporary files. The script counts the number of times the Tab key is pressed with the Alt key held down and waits for the Alt key to be released. If the popup is not running it then also raises the selected window. However if the popup is running, it signals each Tab press and Alt Up to the popup daemon. On first Tab press the popup daemon, creates, displays and with subsequent Tab presses updates a splash list (application name and windows name with selected window in bold) of focusable windows,  then raises the selected window as chosen from the splash after the Alt key is released.

The script is totally reliant on the continuing availability of X11 functionality so should continue to work for the moment but may not survive the eventual move to Wayland. 
**It is to be hoped that we do not move to Wayland before xWayland supports all the commonly used features of X11 including those required by this script**.

I attach the code for the revised script and the popup daemon below as well as the code for some support utilities. The following files are required:

| *Suggested file name*   | *Suggested directory*           | *Function*
| -------------------- | ------------------------------ | ------------------------------------------------------------------------ |
| `alttab.sh`           | `/home/pi/bin/`               | Replacement for Jaciss's script activated by xbindkeys in response to Alt Tab key press |
| `popup.py`            | `/home/pi/bin/`               | Daemon providing popup splash , started via autostart desktop shortcut and script |
| `startpopup.sh`       | `/home/pi/bin/`               | Script to start popup daemon called from desktop shortcut in autostart |
| `startpopup.desktop`  | `/home/pi/.config/autostart/` | Desktop shortcut to autostart daemon |
| `.xbindkeysrc`        | `/home/pi/`                   | xbindkeys configuration file |

The files in `/home/pi/bin` need execute permission. 

The following packages, which may not already be installed, are required and can be installed via `apt install` :

`x11-utils  xbindkeys  xdotool  python3-tk`

Alt Tab needs to be disabled by issuing the following command (not as root)

`gsettings set org.gnome.desktop.wm.keybindings cycle-windows []`

It can be re-enabled, if needs be, by issuing the following command 

`gsettings reset org.gnome.desktop.wm.keybindings cycle-windows`

The popup daemon uses the tkinter graphics library as requirements are undemanding and tkinter is part of the python core language. The daemon displays the popup, with a short delay, the first time the Tab is pressed with Alt key held down with the next item in stacking order preselected. The popup disappears after the user has tabbed through to the required window and released the Alt key. The Alt key needs to be released after the Tab key. If the popup remains (because Alt key was released first), just press and release the Alt key. For toggling between the last two focussed windows a quick Alt Tab and release can be done without waiting for the popup daemon to appear. 

Notes :
* To view the log file for `alttab.sh` , issue the following command in a terminal window:
`tail -f /tmp/tabs.log`
and enable logging in the script by deleting the value `1` in the last but one line of the script
* To view the log file for `popup.py`, disable the popup (if running) by issuing the following 
command in a terminal window
`bash /tmp/killpopupup.sh`
and run `popup.py` from an IDE such as geany
* In running thd in the script, thdevent is set to the value `/dev/input/event0`
because keyboard events came as event0 in my case. If Alt Up detection is not working try with the
value of thdevent changed to `/dev/input/event*`
* If running the script without the popup, note that the existence of any running program with `popup.py` in its name
will prevent the the script from raising the newly required window. In such case kill any running instances of `popup.py`  

If any problems please let me know and I'll try to fix them !

**alttab.sh**

```bash
#!/bin/bash
# 07/03/22 by drdrm46 with many many thanks to Jaciss !
# To be run from xbindkeys with /home/pi/.xbindkeysrc containing :
#       #Alt Tab select and toggle with popup support
#       bash alttab.sh"
#       Alt+Tab
# This script runs every time Tab key is pressed with Alt key held down,
# checks for Alt key up and terminates after Alt key is released or
# timeout. Multiple instances may run simultaneously for a short while.

function execute
{    
    initialise
    count_tabs
    detect_alt_up
    if [ $altup ] ; then
        reset_tab_count
        is_popup_running
[... 364 lines cut ...]
```
**startpopup.sh**
```bash
#!/bin/bash
sleep 10
nohup python3 /home/pi/bin/popup.py & > "/tmp/nohup.out"
pid=$!
echo "kill $pid" >/tmp/killpopup.sh

```
**startpopup.desktop**
```
[Desktop Entry]
Name=Start Popup
Type=Application
Exec=bash /home/pi/bin/startpopup.sh``

```
**.xbindkeysrc**
```bash
# Alt Tab select and toggle with popup
"bash alttab.sh"
Alt+Tab

```