RPi-Distro/raspberrypi-ui-mods #38: Keystroke Repeat Delay & Interval are not respected until lxinput is launched

RaviRJoshi, 2022-02-11:
Steps to Reproduce: 
1. In lxinput, change keyboard delay & interval. 
2. Log Off.
3. Log back in. 
 Note that the delay & intervals behaviour hasn't changed. 

Bug/Feature: It appears that the personalised settings are not loaded when the user logs in. 
Workaround: Just launch 'lxinput', and cancel out.

spl237, 2022-02-11:
Ah - well spotted! I used to have an autostart that ran lxinput at boot time to reload the settings, but it seems to have vanished - I'll try and find out where it went and put it back...

spl237, 2022-02-11:
OK, I know what is going on - the autostart is still there and is still working, but mutter then overrides its settings with its own. There ought to be a way to set the keyboard delay within mutter itself; I'll see if I can work out how to do it.

spl237, 2022-02-11:
Fixed - the settings need to be updated in the gsettings daemon for mutter. I've patched lxinput; the new version is uploading to apt and should be available in the next couple of hours; please update and make sure it fixes the problem for you, and report back here if not.

Many thanks for the report!

closed 2022-02-11 by spl237

RaviRJoshi, 2022-02-11:
Thanks Simon.

RaviRJoshi, 2022-02-16:
Just an observation. 
The fix seems to work only in Beta and not in GA. I've installed both OSes multiple times. Beta is from 10/30/2021, GA is 01/28/22.
LXInput-setup.desktop is present both in beta and GA.

spl237, 2022-02-16:
I'm using it on the latest released image - seems to work for me.
LXInput-setup.desktop is no longer what makes the setting - mutter relies on the gsettings value instead, which is now written by lxinput. This should be identical on the beta and on the released image.