RPi-Distro/raspberrypi-ui-mods #41: Openbox Configuration Manager issue on RaspiOS x86 PC version

b1ack0p, 2022-04-07:
PC: IBM ThinkPad R52
uname -a : Linux <rpiosx86> 4.19.0-20-686-pae #1 SMP Debian 4.19.235-1 (2022-03-17) i686 GNU/Linux

when i get into Applications List > Preferences > Appearance Settings and do some changes like in screenshots attached
https://imgur.com/ciger3M
https://imgur.com/a/fa8KCit

and then go to Applications List > Preferences > Openbox Configuration Manager it changes default PiXflat theme to one of the list, changes my previous settings i did in this tool and **it disables left click mouse function**.

https://imgur.com/ghJKAe5
https://imgur.com/GyeZbrG

spl237, 2022-04-08:
The Openbox Configuration Manager is known not to work in Raspberry Pi OS, which is why we do not show it in the menus. (It shouldn't be shown on the x86 image - if it is, that is a bug.)

Given that openbox is no longer the default window manager, and will be removed from the x86 image when we next update it, I'm afraid this is something we won't fix.

closed 2022-04-08 by spl237

b1ack0p, 2022-04-08:
I understand. When will be the next update for x86? and will there be configuration tools for more customization on theme such as moving close/minimize/maximize buttons to left rather than right. and disable animations.. etc which openbox config manager was doing? and size of the bars ..etc

also could you give me a bit more detail about release date of x64 version? i cant wait to install/test raspiOS on a modern x64 PC.

P.S.: i still think it would be great if Pixel desktop/theme is available externally as a seperate package to install on other distros/desktops.. i can use theme and icons this way on gnome and xfce but not fully supporting. still some icons/colors/items missing..

thanks for this beautiful OS and theme.