RPi-Distro/raspberrypi-ui-mods #42: BUG: new version of raspberrypi-ui-mods needed for buster

theofficialgman, 2022-05-23:
@spl237 
```
Package: raspberrypi-ui-mods
Version: 1.20210706
Architecture: armhf
Maintainer: Simon Long <simon@raspberrypi.com>
```

the buster version lacks the critical update https://github.com/RPi-Distro/raspberrypi-ui-mods/commit/920c8d4d5954b475956f492f99a307b02c078689 that breaks flatpaks and many other functions from showing in applications lists

please update the package for buster armhf/arm64 to include this

spl237, 2022-05-23:
Buster is no longer supported, and will not be updated for anything other than security-critical fixes.

closed 2022-05-23 by spl237

theofficialgman, 2022-05-23:
> security-critical fixes.

this is a security critical fix. in the case where a debian package is out of date, and the user install a flatpak, they will never be able to launch the flatpak from the apps list. leaving the user to continue to launch the outdated debian package which could be vulnerable to fixed security issues in the flatpak

spl237, 2022-05-23:
> this is a security critical fix. in the case where a debian package is out of date, and the user install a flatpak, they will never be able to launch the flatpak from the apps list. leaving the user [...]

It is not a security-critical fix. It may prevent other security-critical fixes from being installed by an installation method which we do not actually support anyway, but that does not make it security-critical itself. It does not prevent security-critical fixes from being installed via apt, which is our supported installation method for updates.

> as buster ages, it will become increasingly critical for users that need to stay on buster to switch over to updated versions of their apps, whether that be through flatpak, snaps, and appimages which overwriting the XDG_DATA_DIRS breaks

We will not be supporting buster other than as an LTS branch, for which our policy is that we take upstream security fixes only. We do not have the resource to continue to support obsolete versions of the OS.

XECDesign, 2022-05-23:
Updating from a source that tracks and fixes security issues (Debian) to one that doesn't (flathub) isn't a security fix.

The point of providing a legacy release is so that people can run the versions of software they're already running.