RPi-Distro/raspberrypi-ui-mods #44: Themes

minimumattic, 2023-06-24:
Hi.
I was wondering if you can help me or at least give me advice. I modify/adaptate your PiX/PiXflat theme for native LXDE/openbox and Xfce on desktop since last year or more. But I noticed some gtk/gnome apps not showing properly.

currently using your theme since Debian 11 and now upgraded to 12 with native LXDE installation. Issues i am having are as following:

such as Context menu arrow icons dont appear in some GTK apps (such as network-manager-gnome gui and Inkscape - if i set theme to system theme/pixflat).  in Debian 11/bullseye i was not having context menu arrow icon problem but problem occured after upgrading to Debian 12/bookworm.

and some gnome apps such as gnome-disk-utility and baobab denying to use PiXflat theme and pushing Adwaita theme. this problem continues since Debian 11.

is there something can be done about Pix and Pixflat themes or can you advice for a fix?

thanks in advance.

minimumattic, 2023-06-24:
Actually context menu arrow icons seems to be working for PiX theme but that theme is a bit older and seems have some broken parts. but context menu arrow icons (in mentioned apps menu items) are not appearing in PiXflat. definitely something feels weird.

minimumattic, 2023-06-24:
Gnome disk utility:
[image]

Gnome Baobab (Disk usage analyzer):
[image]

spl237, 2023-06-24:
Recent GNOME apps pull themeing information from gsettings, not the session daemon. Try installing the dconf-editor app from apt and setting the icon theme and GTK theme in there.

minimumattic, 2023-06-25:
Hi. thanks for replying. i installed dconf-editor but still it doesnt change the theme in there. i tested some other themes in LXDE such as Clearlooks and it applies context menu arrow icons on the GTK apps except PiXflat theme. Maybe PiXflat theme needs a small edit in GTK theme config CSS to pull menu-arrow icon information? PiX theme doesnt have this issue, just PiXflat is doing this annoying issue. dont get why..

spl237, 2023-06-25:
In that case, to be honest, I can't really help you. The themes have been designed and tested to work on RPiOS, and they do - if there is some way in which they are incompatible with other environments, that's outside the scope of anything I have time to look at.

At a guess, you haven't installed the assets subdirectory of the PiXflat theme which stores the graphics you are missing, or it is at a path the theme isn't expecting, something like that - those missing graphics are all stored as png and svg files in the assets directory, so if they aren't being shown, your OS isn't finding them.

minimumattic, 2023-06-25:
I understand. they are both in correct place where themes are usually located ~/.themes or /usr/share/themes

anyway thanks for your time for replying.

closed 2024-05-10 by spl237