RPi-Distro/raspberrypi-ui-mods #45: archlinux install

TamirRothschild, 2023-08-11:
yea i know but you use arch btw, well i cant find any repo/AUR/pkg or someone whom done this before, i really like the DE but the installation process is bad just like arch but worse, can any one point me to the source file so i can compile it or some thing, and whats the apt repo try to install it on a clean debian install but count find the dam repo, i really like your DE thou it lightweight and its built on lxqt.
BTW i use your PC 32bit ver of the raspbian os for my media server is not bad at all very lightweight and easy to maintain.
yes i had to use the BTW is use

spl237, 2023-08-11:
I'm sorry, but I really don't think I can help. There are literally dozens of packages you would need to build for arch and install, and it's not something we are ever going to be able to support.

You could try adding archive.raspberrypi.org to your apt configuration - that is our public repo - and then installing raspberrypi-ui-mods from there, which should then pull in all the rest of the dependencies, but it is still likely to require quite a bit of manual fixup thereafter.

closed 2023-08-11 by spl237

TamirRothschild, 2023-08-12:
its ok i can compile/install pkgs, i guess you have not seen gento yet but 40 pkg as a dependency and probably some of them are already installed.

anyway i really appreciate you responding, thank you.