RPi-Distro/raspberrypi-ui-mods #48: Bad text contrast with new PiXFlat dark theme

Botspot, 2023-12-15:
Problems encountered:
1. URLs are nearly impossible to read.
2. Text on unfocused windows has poor contrast.
[image]
[image]
3. The QT theme matches very well, until you encounter "sort by" bars on lists. Example below is VLC's playlist view. It is impossible to read the white text on the white bar.
[image]
4. I spotted this black text color on `gnome-screenshot`
[image]
5. Very bad contrast on Add/Remove Software program
[image]

Botspot, 2023-12-15:
And overall, I think this rendition of the dark theme is missing out on its true potential. In my opinion it would look much better if large background areas were darkened a bit more.
Example below is Github's builtin dark mode.
[image]
This has very good contrast and is very easy on the eyes, while PiXFlat feels almost foggy or muddy in comparison.

ghollingworth, 2023-12-15:
If you want to propose something different that’s what pull requests are for… If you just want to vent then that’s what the forums are for.

closed 2023-12-15 by ghollingworth

Botspot, 2023-12-15:
> If you want to propose something different that’s what pull requests are for… If you just want to vent then that’s what the forums are for.

That is understandable, but it seemed to me that constructive feedback was requested.
[image]

Botspot, 2023-12-19:
@spl237 Would you be willing to review the text contrast issues identified and check if they are easy to resolve?

ghollingworth, 2023-12-19:
I'm sure he will have a look when he's back from his vacation, but that won't be until January now.

Botspot, 2024-01-01:
6. The buttons and footer on the Advanced Options for the Network plugin have a strange gradient and were clearly not themed this way purposefully.
[image]
7. Bad text contrast when selecting Other Locations on the gtk file chooser.
[image]

Botspot, 2024-01-01:
8. In galculator, the selected calculation modes are displayed in black instead of white. [image]

spl237, 2024-01-02:
1. In which application?
2. Text on background windows is intentionally reduced in contrast, as it is with the light theme. In response to your feedback, I have lightened it slightly, but the intention is that the foreground window should have noticeably more text contrast than other windows.
3. Fixed
4. That text is light-coloured for me, not black. Try updating.
5. Improved as a result of 2.
6. Fixed
7. Fixed
8. Probably hard-coded colour in the calculator app rather than themable, but black is readable against grey anyway, so won't be changed.

Botspot, 2024-01-02:
11. The URL colour in an editable text-info field should be lightened.
    This can be reproduced with a yad dialog.
    Install yad: 
    ```
    sudo apt install yad
    ```
    Then run this simple test window
    ```
    echo -e 'normal text\nhttps://example.com' | yad --text-info --show-uri --width=250
    ```
    Screenshot:
    [image]
    It's possible the URL color is hardcoded into YAD, but any such option is not exposed to the user so I assume the colour is set somewhere in the GTK theme.

spl237, 2024-01-02:
I think you are now mostly into personal preference rather than actual problems. I may revisit some of these at some point in the future, but I need to move onto other things; if people have major contrast issues with the dark theme, then the obvious solution is to use the light theme.

Botspot, 2024-01-02:
12. Bad text contrast on 'choose file' window when unfocused
[image]

Botspot, 2024-01-02:
> I think you are now mostly into personal preference rather than actual problems. I may revisit some of these at some point in the future, but I now need to move onto other things; if you or anyone else still have major contrast issues with the dark theme, then the obvious solution is to use the light theme.

That is fine with me and I understand. For my side, I will continue using the dark theme and continue reporting any newly discovered issues. If for nothing else, this issue could serve as a record of known issues to help prevent duplicates.
IMHO, the only remaining 'major' unresolved issue is the URL color mentioned in 1. Would that be an easy fix?

spl237, 2024-01-02:
>  IMHO, the only remaining 'major' unresolved issue is the URL color mentioned in 1. Would that be an easy fix?

It ought to be, but on the other hand, there is something unusual going on as I have not changed the URL colour from the default blue used in the light theme, and indeed some URLs still display in the correct blue anyway. So I suspect this isn't as simple as URLs are now all the wrong colour, because some of them (including the one in the widget factory tool I use for theme design) are correct. If I get a chance I will have a look now you have given an example I can repeat.

Botspot, 2024-01-02:
> It ought to be, but on the other hand, there is something unusual going on as I have not changed the URL colour from the default blue used in the light theme, and indeed some URLs still display in [...]

Actually I did not give you a simplified reproducing command for the grey url issue, but only for the too dark blue url color for number 11.
Here's a modified adaptation of 11 and 1 that exhibits both bad url colors.
```
echo -e 'normal text\nhttps://example.com' | yad --text-info --show-uri --width=250 --text=$'normal text\n<a href="https://example.com">https://example.com</a>'
```
[image]
Hope that helps.

spl237, 2024-01-03:
OK, URLs should now be fixed - I've just set them to match the colour of other text; I'm not convinced that having URLs in a specific shade of light grey is particularly helpful when it comes to identifying them...

Updated theme should be in apt later on today.

Botspot, 2024-01-03:
> OK, URLs should now be fixed - I've just set them to match the colour of other text; I'm not convinced that having URLs in a specific shade of light grey is particularly helpful when it comes to identifying them...

One is now grey, the other is unchanged. Were you planning to fix both or only the more common kind of URL?
[image]

spl237, 2024-01-03:
I couldn't find where the blue colour of the one below comes from. From my widget toolkits, it appears that it was actually GTK2 that used that colour for links - GTK3 uses the grey - but I couldn't find anything in the GTK2 theme files to change it. How yad is pulling colours from both GTK2 and GTK3 on the same window is a puzzle though - albeit not one I have time to investigate at the moment...