RPi-Distro/raspberrypi-ui-mods #49: More contrast issues with PiXFlat dark

Botspot, 2024-01-24:
Trying to keep this to actual contrast issues for @spl237 to look at.

1. Evince zoom level drop-down arrow: 
Normal:
[image]
Mouse over: (mouse is invisible in grim screenshot)
[image]
I think the arrow should be white all the time.
I am unfamiliar with Evince and actually spent a few moments looking for a way to change the zoom level as I did not notice the grey arrow.
2. Evince floating drop-down list improperly highlights the item under the mouse:
[image]
Probably the most cohesive thing to do here is to make the button background color a bit lighter grey when under the mouse pointer.
3. Calendar arrows (to change the month and the year) are grey in normal state and white when under the mouse:
[image]
They should probably be white at all times.

Botspot, 2024-03-24:
5. Line numbers on Mousepad unreadable.
[image]

theofficialgman, 2024-05-30:
> 5. Line numbers on Mousepad unreadable.

I think this is a mousepad bug where it hardcodes that light grey background and does not read from the gtk theme.