RPi-Distro/raspberrypi-ui-mods #50: Incorrect arrow icon on horizontal scrollbars

Botspot, 2024-05-10:
I think I would have spotted this years ago if it was always an issue, but it has been this way for most of 2024 at a minimum.

Encountered in Geany:
[image]
Encountered in Mousepad:
[image]
Encountered in lxtask:
[image]
Notably, this seems tied to the GTK theme as QT windows and GTK4 windows are unaffected.

spl237, 2024-05-10:
Weird. I changed those icons years ago, but not recently, and they were definitely correct when I changed them! But it looks as if something has got out of whack somewhere - I'll have a look next week.

spl237, 2024-05-10:
Ah - I see the problem; there's an incorrect icon reference in the GTK 3 theme CSS file. Must have accidentally hit the wrong button at some point when editing something else in the file and not noticed the one letter change! Will fix now - should be in apt at some point over the next few days.

spl237, 2024-05-10:
Fixed - was a cut-and-paste error when I was tidying the handling of symbolic icons while creating the dark theme. Well spotted!

Should be in apt by Monday.

closed 2024-05-10 by spl237